ExpeditionExploration/opengpio #1: Feature Request: Support Pull-Up and Pull-Down settings

Tom-Hirschberger, 2024-03-26:
Hi, switched to your library recently as Kernel 6.6 broke one of my MagicMirror modules.
Would be grate to be able to set pull-up or pull-down for pins while registering them.

kueckermann, 2024-04-09:
Having some issues with this one. It seems that while libgpiod does have the functionality for pull up/down and debounce, it doesn't seem to be exposed in the development headers from libgpiod-dev. The headers for the core cpp library do clearly have these parameters available. You'll notice in this file that internal bias for pull up/down is there: https://github.com/brgl/libgpiod/blob/master/bindings/cxx/gpiodcxx/line.hpp

But the header file that comes from libgpiod-dev doesn't have any availability for this, as reference here the section from the header file for line: 
```cpp
class line
{
public:

	/**
	 * @brief Default constructor. Creates an empty line object.
	 */
	GPIOD_API line(void);

	/**
	 * @brief Copy constructor.
	 * @param other Other line object.
	 */
	GPIOD_API line(const line& other) = default;

	/**
	 * @brief Move constructor.
	 * @param other Other line object.
	 */
	GPIOD_API line(line&& other) = default;

	/**
	 * @brief Assignment operator.
	 * @param other Other line object.
	 * @return Reference to this object.
	 */
	GPIOD_API line& operator=(const line& other) = default;

	/**
	 * @brief Move assignment operator.
	 * @param other Other line object.
	 * @return Reference to this object.
	 */
	GPIOD_API line& operator=(line&& other) = default;

	/**
	 * @brief Destructor.
	 */
	GPIOD_API ~line(void) = default;

	/**
	 * @brief Get the offset of this line.
	 * @return Offet of this line.
	 */
	GPIOD_API unsigned int offset(void) const;

	/**
	 * @brief Get the name of this line (if any).
	 * @return Name of this line or an empty string if it is unnamed.
	 */
	GPIOD_API ::std::string name(void) const;

	/**
	 * @brief Get the consumer of this line (if any).
	 * @return Name of the consumer of this line or an empty string if it
	 *         is unused.
	 */
	GPIOD_API ::std::string consumer(void) const;

	/**
	 * @brief Get current direction of this line.
	 * @return Current direction setting.
	 */
	GPIOD_API int direction(void) const noexcept;

	/**
[... 113 lines cut ...]
	 */
	GPIOD_API bool operator!(void) const noexcept;

	/**
	 * @brief Possible direction settings.
	 */
	enum : int {
		DIRECTION_INPUT = 1,
		/**< Line's direction setting is input. */
		DIRECTION_OUTPUT,
		/**< Line's direction setting is output. */
	};

	/**
	 * @brief Possible active state settings.
	 */
	enum : int {
		ACTIVE_LOW = 1,
		/**< Line's active state is low. */
		ACTIVE_HIGH,
		/**< Line's active state is high. */
	};
```

Not sure why, but can only think that this is newer functionality that isn't exposed in libgpiod-dev yet. But that's just a guess. Will see if it's possible to directly reference the header files from the core library instead of from libgpiod-dev.

Tom-Hirschberger, 2024-04-14:
Hi,

I do not have much knowledge of C++ but it looks like the last changes of the bindings are at least two years old.

https://github.com/brgl/libgpiod/blob/master/bindings/cxx/line.cpp

Thought it might be cause of the change to version 2 of the lib but the bias and pull settings are included into the 1.6.x branch, too.

kueckermann, 2024-04-15:
I noticed found devs that seemed to be working on dev bindings for version 2, which is why I also suspected that the current dev bindings were for an older version. I'll try include version 2 as a git submodule and use the header files directly from there. I'm traveling for work this week, so I'll probably only get onto this again next week.

kueckermann, 2024-04-24:
@Tom-Hirschberger I'm back from my work trip and I've had a look into this. I was able to add libgpiod as a submodule to the package. Thankfully it doesn't require any external dependancies so the project build pipeline works just fine. However, the version 2.1.x structure is not compatible with the headers exposed by libgpiod 1.6.x, so the compiler throws a bunch of errors. 

Thankfully tho, version 2 seems relatively straight forward so I should be able to reuse the majority of the existing opengpio.cpp code with changes mostly around how c++ requests the chip and line objects. I was able to get a striped down version of the file to compile.

I'm working on this on the branch **version/libgpiod_v2.1.x**. It's probably going to take a bit of work still, but once done, we'll be able to make use of all the features within version 2 of libgpiod. I'll keep you updated.

Tom-Hirschberger, 2024-04-25:
Thank you very much for your affort.

kueckermann, 2024-04-27:
@Tom-Hirschberger managed to get it working for input, output, and watching. Debounce and Bias appear to be working. I need to finish up PWM still. The only thing that I need to really work on setting up is to automate the download, build, and install of libgpiod v2.1.x. I can't find a prebuilt version of this that can be installed via apt install, so for now it seems that the easiest is that it builds and installs on the device from the source code. This seems fine because it doesn't require any dependancies, but that downside is it takes several minutes to install. In future, rebuilt binaries could be packaged with the library.

If you want to try it out, you can do the following:
- Clone libgpiod from it's repo to your device using the version 2 branch. 
`git clone -b v2.1.x git@github.com:brgl/libgpiod.git`
- cd into the repo and run these 3 commands to build it:
```
./autogen.sh --enable-bindings-cxx
make
make install
```
This should build and install the latest version of libgpiod to your device. 

- Now you should be fine to npm install the work in progress version of the library using: 
`npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"`

Nothing changes in the library, only input and watch methods now support an optional options object where you can set debounce (for watch) and bias (for watch or input).

Here's an example:

```
import { NanoPi_NEO3, Edge, Bias } from '../src/';

const watch = NanoPi_NEO3.watch(NanoPi_NEO3.bcm.GPIO2_B7, Edge.Both, {
    debounce: 20,
    bias: Bias.PullDown
});
```

If you get. a chance to try this, let me know how it goes and what issues you encounter.

kueckermann, 2024-04-28:
@Tom-Hirschberger I've completed the setup of the library. 
Would you mind to give it a try? You can install using `npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"`

Note that it will take a while to install since it's building from source code and you'll probably think the install has jammed, but just let it be. Even for like 15 minutes.  If this works, I'll probably look at packaging the prebuild binaries atleast for ARM.

kueckermann, 2024-04-29:
@Tom-Hirschberger no stress on timing. I've just pushed a change that includes the bindings for ARM so it doesn't need to rebuild. Are you using an ARM based system?

kueckermann, 2024-04-29:
@Tom-Hirschberger no stress. Are you using an ARM based system?

kueckermann, 2024-04-30:
@Tom-Hirschberger after much trial and error I opted to remove auto building of libgpiod version 2 from being embedded in the library. Rather, I added instructions on how to build the prerequisites into the readme. It's easy enough for people to do anyway. 

I've pushed the latest version to: https://github.com/ExplorationSystems/opengpio/tree/version/libgpiod_v2.1.x
You can follow the instructions in the README to install libgpiod v2.1.

After you've installed it successfully, you can run `npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"` to install the library. 

Everything should work as before, expect there are new additions to use the built in debounce and bias.
Here's an example:
```js
import { NanoPi_NEO3, Edge, Bias } from 'opengpio';

const watch = NanoPi_NEO3.watch(NanoPi_NEO3.bcm.GPIO2_B7, Edge.Both, {
    debounce: 20,
    bias: Bias.PullDown
});
```

No rush, but once you've given this new version a try and can confirm that it works then I'll pull it into main and deploy it to npm as a new version.

kueckermann, 2024-04-30:
I've also setup a script to automate the install process for libgpiod, you can execute the following to use it:
`curl https://raw.githubusercontent.com/ExplorationSystems/opengpio/version/libgpiod_v2.1.x/install-libgpiod.sh | bash`

Tom-Hirschberger, 2024-05-09:
Hi, had some minutes and tried the installation.
The script quits with an error during compilation of the library.

```
pi@mirror-dev:~ $ curl https://raw.githubusercontent.com/ExplorationSystems/opengpio/version/libgpiod_v2.1.x/install-libgpiod.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   522  100   522    0     0   1955      0 --:--:-- --:--:-- --:--:--  1962
Cleaning directory
Ensure build dependancies
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
tar ist schon die neueste Version (1.34+dfsg-1.2+deb12u1).
gzip ist schon die neueste Version (1.12-1).
build-essential ist schon die neueste Version (12.9).
autoconf ist schon die neueste Version (2.71-3).
curl ist schon die neueste Version (7.88.1-10+deb12u5).
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libcamera0.1 libssl1.1 linux-headers-6.1.0-rpi7-common-rpi
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 104 nicht aktualisiert.
Fetching libgpiod v2.1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  206k    0  206k    0     0   560k      0 --:--:-- --:--:-- --:--:--  562k
Build and install libgpiod
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: creating directory autostuff
autoreconf: configure.ac: not using Libtool
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /usr/bin/autoconf --force
configure.ac:75: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:75: You should run autoupdate.
./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:75: the top level
configure.ac:203: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: error: /usr/bin/autoconf failed with exit status: 1
make: *** Es wurden keine Ziele angegeben und keine „make“-Steuerdatei gefunden.  Schluss.
make: *** Keine Regel, um „install“ zu erstellen.  Schluss.
```

kueckermann, 2024-05-09:
@Tom-Hirschberger thanks for checking. Seems like a build dependency is missing, I'll add it to the script.

kueckermann, 2024-05-13:
@Tom-Hirschberger could you try two things:

1. I've added an additional build lib that seems to be missing based on the logs you provided. Can you try run `curl https://raw.githubusercontent.com/ExplorationSystems/opengpio/version/libgpiod_v2.1.x/install-libgpiod.sh | bash` again and see what happens? I feel this approach might be a big of trial and error tho. 

2. If the above still fails can you try following the build script at the lib's readme? https://github.com/brgl/libgpiod/tree/v2.1.x . If you encounter any missing libs, try install them and let me know which ones were missing. I'll add them to the automated script.

kueckermann, 2024-05-13:
Last thing... What version of linux are you on?

Tom-Hirschberger, 2024-05-13:
Hi, I had the plan to debug further to provide more information to you already but had not found any time yet. But I will try to in the next days.

I am running two systems at the moment. One is Debian Bullseye 32bit, one is Debian Bookworm 64bit.

kueckermann, 2024-05-13:
No stress @Tom-Hirschberger. Also been very busy my side the last week. 👍

Tom-Hirschberger, 2024-05-20:
Hi,

i had to modify your installer slightly:

```bash
# Clean directory
echo "Cleaning directory"
rm -rf libgpiod-2.1

#autoupdate package does not exist
#
echo "Update apt cache"
sudo apt -y update

# Ensure build dependancies
echo "Ensure build dependancies"
sudo apt -y install tar gzip build-essential autoconf curl autoconf-archive libtool

# Fetch libgpiod on branch 2.1.x
echo "Fetching libgpiod v2.1"
curl -o libgpiod-2.1.tar.gz 'https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/snapshot/libgpiod-2.1.tar.gz'
tar xf libgpiod-2.1.tar.gz
cd libgpiod-2.1

# Build and install libgpiod
echo "Build and install libgpiod"
./autogen.sh --enable-bindings-cxx
make
sudo make install
```

* do a `apt update` before the apt install
* added `-y` option to apt commands
* `autoupdate` package did not exist so i removed it of apt install
* added `libtool` to apt install
* added `sudo` to make install

With the `libtool` dependency installed the lib gets compiled and is installed but although i removed my `node_modules` directory and the `package-lock.json` before re-run the `npm install` i get the following error if  run my test script:

```bash
node index.js 
node: symbol lookup error: /home/pi/test/node_modules/opengpio/build/Release/opengpio.node: undefined symbol: _ZN5gpiod13line_settingsC1Ev
```

Tried to run `sudo ldconfig` after the compile to make sure the system recognized the new lib but it does not make any difference.

kueckermann, 2024-05-20:
That's super helpful. Thanks a ton. 

The linker issue is an odd one. I'll take a look into this. The only thing that initially comes to mind is if it's somehow utilising a file from the old version of libgpiod. I think I'll try install debian on my device and run this to see if I can reproduce the linker issue. 

I might send a few questions but I think this sufficient information for me to dig further.

srl295, 2024-08-16:
ANd I meant to report back - pull up and debounce settings work great for me! 

https://github.com/codehivetx/barrelcactus-watermeter/blob/520c11af53ad0e8dd3aaedd24a156a062f66ad21/src/gpio/tankgpio.ts#L173-L177

kueckermann, 2024-08-21:
@srl295 super expecting to hear that it's working for you! I'm really glad others are also finding this library valuable. Thanks for sharing your project. 

I need to do a bit more finalisation work on this version of the library but hoping to release as a beta version to npm sometime in September. Thanks also for the PR, it's merged.

Grothesk242, 2024-08-25:
Instead of building it from source I did cherry pick libgpiod3 from Debian Experimental: 
Will opengpio work with that installation?
```
libgpiod3:
  Installed: 2.1.2-1~exp1+b1
  Candidate: 2.1.2-1~exp1+b1
  Version table:
 *** 2.1.2-1~exp1+b1 100
          1 https://deb.debian.org/debian experimental/main arm64 Packages
        100 /var/lib/dpkg/status
libgpiod-dev:
  Installed: 2.1.2-1~exp1+b1
  Candidate: 2.1.2-1~exp1+b1
  Version table:
 *** 2.1.2-1~exp1+b1 100
          1 https://deb.debian.org/debian experimental/main arm64 Packages
        100 /var/lib/dpkg/status
     1.6.3-1.1+b1 500
        500 http://deb.debian.org/debian trixie/main arm64 Packages
```

kueckermann, 2024-08-25:
@Grothesk242 if the libgpiod version is correct then this new version should work with it as it should use the header files for what ever version of libgpiod is installed on your system. However, you may encounter issues if you've previously had an older version of libgpiod installed. 

You can try install libgpiod as you've mentioned here, and then install the new version of the library from the branch using `npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"`

Let me know what you experience. 

Can you also share the command you used to install the libgpiod3 package?

Grothesk242, 2024-08-25:
Add this entry to /etc/apt/sources.list
```
deb https://deb.debian.org/debian experimental main
```
Then you can install gpiod/libgpiod3 the usual way: 

```
apt update
apt -t experimental install gpiod libgpiod3  libgpiod-dev
```

kueckermann, 2024-08-30:
Thanks @Grothesk242 I'll try this over the weekend and get back to you.

Grothesk242, 2024-08-30:
Just for your information, the package left 'experimental' and is now available in 'unstable' repository.

Tom-Hirschberger, 2024-12-17:
Hi, my year had been busy very much, too. The current implementation worked well for most of my scenarios and had no time to test something new.
My twins kept me busy most of the time.

Thank you for still trying to implement the features.

MeestorX, 2025-01-14:
> ANd I meant to report back - pull up and debounce settings work great for me! in production.
> 
Came here looking for the pull-up and pull-down functionality. It sounds like it's possible now? Would someone be able to make some TL;DR instructions for how to use it?

kueckermann, 2025-01-15:
Hey @MeestorX. Here's the TL;DR. Also note myself and @jtholmi are actively working on this new version at the moment. The primary reason it's taking time is because of linker issues if people have previously installed libgpiod.

If you run into an issue when installing directly from this branch then you'll likely need to purge libpgiod version 1 and reinstall it. I'll link some code to do that.

You can try installing directly from this branch using: `npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"`

Note this builds the latest version of libgpiod from source. If you get errors saying some library is missing then install the missing library manually. There is already a script which attempts to install the relevant libraries. 

If you get a `symbol lookup error` when running your project then it's likely because the linker has linked to version 1 of libgpiod which you have installed. To fix this. You can run:
```
rm -rf /usr/include/gpiod*
rm -rf /usr/local/include/gpiod*
```
Then you can reinstall the library using the npm install command above. If you still get an issue you can try run `sudo ldconfig`.

If you are concerned about what will get deleted running rm, you can first run:
```
ls /usr/include/gpiod*
ls /usr/local/include/gpiod*
```
To sanity check what will be removed.

Let me know how you get along.

MeestorX, 2025-01-15:
Awesome. Thank you for replying so quickly! This library has saved my life, after trying unsuccessfully to get pigpio, mraa, etc. to work under Armbian on a Rock S0, so thank you to everyone working on this.

I see that in the later builds, the Radxa_Rock_s0 device is no longer there. I assume it's still supported, if I manually use the `{ chip: x, line: y }` syntax?

kueckermann, 2025-01-15:
Thanks for the great feedback @MeestorX. You're right it won't be on this branch since this is currently not up to date to main. I'm traveling at the moment but will rebase this PR to main this evening and then the Radax device should be there.

MeestorX, 2025-01-15:
Cool. At least 1 pin template (PIN_16 IIRC) was missing on the Radxa, so I can open a new issue to address that after I check to see if there's other ones missing as well.

kueckermann, 2025-01-15:
@MeestorX you can try reinstall the branch now. It's been rebased. Unfortunately Im not at an embedded device that I can test that rebase on, so let me know if you encounter any odd issues.

MeestorX, 2025-01-15:
Added "issue" #22

MeestorX, 2025-01-15:
Looks like I missed a step? I deleted the folders you suggested and ran ldconfig.
```
yarn add "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"
yarn add v1.22.22
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Building fresh packages...
error /workspaces/RotoRod/node_modules/opengpio: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments: 
Directory: /workspaces/RotoRod/node_modules/opengpio
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@11.0.0
gyp info using node@22.13.0 | linux | x64
gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"

gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args '/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args 'binding.gyp',
gyp info spawn args '-f',
gyp info spawn args 'make',
gyp info spawn args '-I',
gyp info spawn args '/workspaces/RotoRod/node_modules/opengpio/build/config.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/vscode/.cache/node-gyp/22.13.0/include/node/common.gypi',
gyp info spawn args '-Dlibrary=shared_library',
gyp info spawn args '-Dvisibility=default',
gyp info spawn args '-Dnode_root_dir=/home/vscode/.cache/node-gyp/22.13.0',
gyp info spawn args '-Dnode_gyp_dir=/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args '-Dnode_lib_file=/home/vscode/.cache/node-gyp/22.13.0/<(target_arch)/node.lib',
gyp info spawn args '-Dmodule_root_dir=/workspaces/RotoRod/node_modules/opengpio',
gyp info spawn args '-Dnode_engine=v8',
gyp info spawn args '--depth=.',
gyp info spawn args '--no-parallel',
gyp info spawn args '--generator-output',
gyp info spawn args 'build',
gyp info spawn args '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/workspaces/RotoRod/node_modules/opengpio/build'
  CXX(target) Release/obj.target/opengpio/cpp/opengpio.o
../cpp/opengpio.cpp:3:10: fatal error: gpiod.hpp: No such file or directory
    3 | #include <gpiod.hpp>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [opengpio.target.mk:110: Release/obj.target/opengpio/cpp/opengpio.o] Error 1
make: Leaving directory '/workspaces/RotoRod/node_modules/opengpio/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack at ChildProcess.<anonymous> (/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:216:23)
gyp ERR! System Linux 6.10.14-linuxkit
gyp ERR! command "/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/bin/node" "/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /workspaces/RotoRod/node_modules/opengpio
gyp ERR! node -v v22.13.0
gyp ERR! node-gyp -v v11.0.0
gyp ERR! not ok
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.
```

kueckermann, 2025-01-15:
@MeestorX can you try with npm instead of yarn? I'm not sure what the behaviour of yarn is when it comes to preinstallation scripts but it seems like it's not running the preinstallation

MeestorX, 2025-01-15:
I don't use npm but if I can't figure out I'll try it. Is node 22 supported?

MeestorX, 2025-01-15:
Same error with npm.
```
vscode ➜ /workspaces/RotoRod $ npm install "https://github.com/ExplorationSystems/opengpio.git#version/libgpiod_v2.1.x"
npm error code 1
npm error path /workspaces/RotoRod/node_modules/opengpio
npm error command failed
npm error command sh -c node-gyp rebuild
npm error make: Entering directory '/workspaces/RotoRod/node_modules/opengpio/build'
npm error   CXX(target) Release/obj.target/opengpio/cpp/opengpio.o
npm error make: Leaving directory '/workspaces/RotoRod/node_modules/opengpio/build'
npm error gyp info it worked if it ends with ok
npm error gyp info using node-gyp@11.0.0
npm error gyp info using node@22.13.0 | linux | x64
npm error gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"
npm error gyp info spawn /usr/bin/python3
npm error gyp info spawn args [
npm error gyp info spawn args '/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm error gyp info spawn args 'binding.gyp',
npm error gyp info spawn args '-f',
npm error gyp info spawn args 'make',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/workspaces/RotoRod/node_modules/opengpio/build/config.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/home/vscode/.cache/node-gyp/22.13.0/include/node/common.gypi',
npm error gyp info spawn args '-Dlibrary=shared_library',
npm error gyp info spawn args '-Dvisibility=default',
npm error gyp info spawn args '-Dnode_root_dir=/home/vscode/.cache/node-gyp/22.13.0',
npm error gyp info spawn args '-Dnode_gyp_dir=/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp',
npm error gyp info spawn args '-Dnode_lib_file=/home/vscode/.cache/node-gyp/22.13.0/<(target_arch)/node.lib',
npm error gyp info spawn args '-Dmodule_root_dir=/workspaces/RotoRod/node_modules/opengpio',
npm error gyp info spawn args '-Dnode_engine=v8',
npm error gyp info spawn args '--depth=.',
npm error gyp info spawn args '--no-parallel',
npm error gyp info spawn args '--generator-output',
npm error gyp info spawn args 'build',
npm error gyp info spawn args '-Goutput_dir=.'
npm error gyp info spawn args ]
npm error gyp info spawn make
npm error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm error ../cpp/opengpio.cpp:3:10: fatal error: gpiod.hpp: No such file or directory
npm error     3 | #include <gpiod.hpp>
npm error       |          ^~~~~~~~~~~
npm error compilation terminated.
npm error make: *** [opengpio.target.mk:110: Release/obj.target/opengpio/cpp/opengpio.o] Error 1
npm error gyp ERR! build error 
npm error gyp ERR! stack Error: `make` failed with exit code: 2
npm error gyp ERR! stack at ChildProcess.<anonymous> (/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:216:23)
npm error gyp ERR! System Linux 6.10.14-linuxkit
npm error gyp ERR! command "/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/bin/node" "/home/vscode/.local/share/fnm/node-versions/v22.13.0/installation/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm error gyp ERR! cwd /workspaces/RotoRod/node_modules/opengpio
npm error gyp ERR! node -v v22.13.0
npm error gyp ERR! node-gyp -v v11.0.0
npm error gyp ERR! not ok
npm notice
npm notice New major version of npm available! 10.9.2 -> 11.0.0
npm notice Changelog: https://github.com/npm/cli/releases/tag/v11.0.0
npm notice To update run: npm install -g npm@11.0.0
npm notice
npm error A complete log of this run can be found in: /home/vscode/.npm/_logs/2025-01-15T23_11_17_854Z-debug-0.log
```

MeestorX, 2025-01-15:
Trying some chatGPT suggestions, but it's not getting me anywhere. It's suggesting I rebuild libgpio from scratch, but the instructions it gives are wrong.
I'll leave this for now and come back to it.

srl295, 2025-01-16:
i think this (npm ) module is trying to use the script here -
https://github.com/ExplorationSystems/opengpio/blob/version/libgpiod_v2.1.x/install-libgpiod.sh
perhaps you could try running that by itself (in a separate directory)

kueckermann, 2025-05-05:
Version 2 is now released (finally). Can be used with `npm install opengpio@latest`.
Will be closing this issue now but feel free to open any new issues as they arise.

closed 2025-05-05 by kueckermann