ExpeditionExploration/opengpio #2: Feature Request: Support setting debounce value

Tom-Hirschberger, 2024-03-26:
As I wrote in the other issue I switched to your lib.
Would be grate to be able to use the debounce settings of libgpiod.

kueckermann, 2024-03-27:
Would you be able to link to the documentation for the debounce of the lib you were using? Just so I can look back at it as a reference?

Tom-Hirschberger, 2024-03-27:
I used to use the "onoff" lib https://github.com/fivdi/onoff.
But at it uses deprecated /sys/class/gpio it uses epoll lib to handle the debounce.

I look at the code of libgpiod (can read C++ but wrote at least 10 years ago the last time) and i think i saw debounce handling already.
So maybe it is only a call of a system setter function (https://github.com/brgl/libgpiod/blob/02e05a134f3c720cb3adcc2aa10e7478dc00f72f/lib/line-config.c#L317)

kueckermann, 2024-05-13:
Being addressed along with issue #1

kueckermann, 2025-05-05:
@Tom-Hirschberger Version 2 is released, and with that the C code can now support debounce. I've started working on adding this in but it's not complete yet.