ExpeditionExploration/opengpio #3: Feature Request: Support reading of current value of a watched pin

Tom-Hirschberger, 2024-03-26:
In my routine of handling a rotary encoder I need to watch two pins. If one pin gets triggered I need to quiry the current value of the other pin.
Do you think it would be possible to support reading the current value of a pin that is already watched?

kueckermann, 2024-03-27:
Good question. My gut feel is it should be possible to add it, although it would likely need to be somehow built into the watcher class. I suspect that if you try create a watcher and reader on the same pin it will crash, as you have discovered in the other issue. It should be possible in the c++ watcher thread to allow some callback to get the current value on the existing open pin. However, I do expect some hidden complexity here since the watcher is running in a separate thread.

Tom-Hirschberger, 2024-03-27:
In my old implementation i did something like this:

```js
const OpenGPIO = require("opengpio")
const openGPIOChip = OpenGPIO.Default
const openGPIOEdge = OpenGPIO.Edge

let curGPIOObj = {chip: 0, line:12}
let curGPIOObj2 = {chip: 0, line: 13}

try{
    let watch = openGPIOChip.watch(curGPIOObj, openGPIOEdge.Both)
    let watch2 = openGPIOChip.watch(curGPIOObj2, openGPIOEdge.Both)

    watch.on('event', (value) => {
        console.log("Watched pin one got triggered with value "+value+"!");
        if (watch2.currentValue){
            //do some crazy stuff
        } else {
            //do some other crazy stuff
        }
    })

    watch2.on('event', (value) => {
        console.log("Watched pin two got triggered with value "+value+"!");
        if (watch.currentValue){
            //do some crazy stuff
        } else {
            //do some other crazy stuff
        }
    })
    console.log("Successfully registered pins")
} catch {
    console.log("Could not register pins!")
}
```

Tom-Hirschberger, 2024-03-27:
My problem was that both pins of the rotary could change their value if you turn it. Depending which one is the  first the rotary is turned clockwise or counterclockwise.
Found a different example which works by only watching one pin and defining the other one as input and it is working for the moment but there might be situations where it is useful.

kueckermann, 2024-04-01:
@Tom-Hirschberger I've tagged you on a PR for this change: https://github.com/ExplorationSystems/opengpio/pull/7

Would you mind giving it a test and letting me know if it works for you as expected?

closed 2024-04-02 by kueckermann

kueckermann, 2024-04-02:
@Tom-Hirschberger deployed to npm in version 1.0.13