ExpeditionExploration/opengpio #4: Feature Request: Exception Handling

Tom-Hirschberger, 2024-03-26:
If I try to register a pin that is already used by a different app or which is not existend my whole node application quits.
I tried to catch the exception but I am unable to.
Can you support wrapping the exception into a less severe one I can catch in my code and do a error handling in the way I like.

Ps.: Sorry for the big amount of issues. I thought it might be helpful to split to different tickets so you can decide for each one individually

kueckermann, 2024-03-27:
Hey @Tom-Hirschberger thanks for the requests. I'll try my best to get to them in a timely manner (just had a baby so time seems to dissappear).
A few questions on this one specifically.
1. What device/operating system are you running on currently?
2. Could you possible share a code snippit that I can use to try reproduce the exception?

Tom-Hirschberger, 2024-03-27:
Hi and thank you for your quick response.
Got twins 11 weeks ago. I can feel you ;-)

I am running two Raspberry 4. One is setup with Bullseye 32bit the other one with Bookworm 64bit. I will try create a pull request to add Raspberry 4 to your devices in the next days. Do have some Raspberry Zero, Zero 2 and Raspberry 3B here as well. Maybe i can add them, too.

I will try to create a simple NodeJS-Test-App to reproduce the problem as my module (https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications) only runs if you install the whole MagicMirror application (https://github.com/MagicMirrorOrg/MagicMirror).

Tom-Hirschberger, 2024-03-27:
The application currently uses Node.js in version v20.11.1

index.js
```js
const OpenGPIO = require("opengpio")
const openGPIOChip = OpenGPIO.Default
const openGPIOEdge = OpenGPIO.Edge

let curGPIOObj = {chip: 0, line:12}

try{
    let watch = openGPIOChip.watch(curGPIOObj, openGPIOEdge.Both)

    watch.on('event', (value) => {
        if (value){
            value = 1
        } else {
            value = 0
        }
        console.log("Watched pin got triggered with value "+value+"!");
    })
    console.log("Successfully registered pin")
} catch {
    console.log("Could not register pin!")
}
```

package.json
```json
{
  "name": "test",
  "version": "0.0.1",
  "description": "Basic test app to reproduce exception in OpenGPIO",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "dependencies": {
    "opengpio": "latest"
  },
  "author": "",
  "license": "ISC"
}
```

run:
```
node index.js
```

If i run a second instance of the app or any other program already uses the pin the app crashes with this message (although there is a catch block):
```
terminate called after throwing an instance of 'std::system_error'
  what():  error requesting GPIO lines: Device or resource busy
Aborted
```

kueckermann, 2024-03-27:
Thanks for this. Definitely some exception handling nessesary in the c++ addon. Native addons can be quite unforgiving at times 😅

kueckermann, 2024-03-27:
Would greatly appreciate if you can do a pull request to add the Raspberry PI devices to the official devices list when you get time 🙏

kueckermann, 2024-04-02:
@Tom-Hirschberger deployed in version 1.0.13