ExpeditionExploration/opengpio #6: Replace console.log with debug lib

kueckermann, 2024-04-01:
Currently since this library is quite early in its development there are several console.log statements. This should be replaced with a namespaced debug logger from https://www.npmjs.com/package/debug

Suggested namespace: opengpio

paony, 2024-12-10:
please fix this urgently, especially remove

console.log('Rising Event', value);
console.log('Falling Event', value);
from Watch.ts

They are annoying, because I get plenty of those events and my own console.log's are hard to find between all those messages

kueckermann, 2024-12-17:
Thanks @paony I'll aim to fix this, this week. I've had a crazy busy year end and haven't had a moment to get to these issue. I appreciate the ping.

paony, 2024-12-18:
No worries, for the moment I patched that locally, but I think it should be fixed anyway and it schould be easy, would be sufficient to comment it out for the moment

closed 2024-12-20 by kueckermann