ExpeditionExploration/opengpio #10: Raspberry Pi 5, problem when using PWM

benjaminfrancois, 2024-05-08:
Hello, 

I want to use PWM with GPIO 18 on RPI 5, but I get this error 

```bash
what():  error setting GPIO line values: Invalid argument
```

This is the code :
```ts
    const pwm = RaspberryPi_5B.pwm(RaspberryPi_5B.bcm.GPIO18, 0, 10)
```

I have search on opengpio.cpp : 
```cpp

    try
    {
        line.request({resourceName, gpiod::line_request::EVENT_BOTH_EDGES, 0}, 0);
    }
    catch (const std::exception &e)
    {
        Napi::Error error = Napi::Error::New(info.Env(), e.what());
        error.ThrowAsJavaScriptException();
        return Napi::Array::New(info.Env());
    }
```

kueckermann, 2024-05-13:
Thanks @benjaminfrancois, I'll take a look into this. It seems like the driver values for RaspberryPi_5B.bcm.GPIO18 are incorrect. Can you try using a different pin and let me know if the same issue occurs? I want to know if it's a problem with all the pin values or just this specific one.

kueckermann, 2024-05-13:
@Tom-Hirschberger would you possibly be able to test this pin?

Tom-Hirschberger, 2024-05-13:
Hi, I will try to test but I can not promise it will be in the next days.

Tom-Hirschberger, 2024-05-20:
Hi,

i run the following test today on both my Raspberry Pi 4 and 5:

```
const OpenGPIO = require("opengpio")
const openGPIOChip = OpenGPIO.RaspberryPi_4B
const openGPIOEdge = OpenGPIO.Edge

try{
    let watch1 = openGPIOChip.watch(openGPIOChip.bcm.GPIO6, openGPIOEdge.Both)
    let watch2 = openGPIOChip.watch(openGPIOChip.bcm.GPIO5, openGPIOEdge.Both)

    watch1.on('event', (value) => {
        console.log("Watched pin 1 got triggered with value "+value+"!");
        console.log("Pin 2 has value: "+watch2.value)
    })
    watch2.on('event', (value) => {
        console.log("Watched pin 2 got triggered with value "+value+"!");
        console.log("Pin 1 has value: "+watch1.value)
    })
    console.log("Successfully registered pin")
} catch {
    console.log("Could not register pin!")
}

const pwm = openGPIOChip.pwm(openGPIOChip.bcm.GPIO18, 0, 10)
```

I used the lastest public npm release of the lib.

But get the same result on both Pis:

```bash
Successfully registered pin
Pwm GPIO at: 0:18 -> 0.000000:10
terminate called after throwing an instance of 'std::system_error'
  what():  error setting GPIO line values: Invalid argument
Aborted
```

The "normal" input GPIOs work as expected but PWM seems broken.

kueckermann, 2024-05-20:
Yea it should work since it's software driven. But it's very helpful to know it's specifically throwing on the PWM.

- Do you get an error if you try using GPIO18 for watch, input, or output?
- Do you get an error doing PWM on GPIO6?

Do you know if the hardware PWM pins need to be enable via raspi-config? I wonder if they have been somehow hardware configured, if that imposes a limitation on the pins to be utilised as IO pins.

Tom-Hirschberger, 2024-05-20:
Hi,
check the Raspberry doc but could not find any hint to disable hardware pwm on the used pins.

I tried to set mode to output in the /boot/firmware/config.txt but this did not change anything.

I can use GPIO18 as input, output and watch pin without any problems.

The same error occurs if i try to use GPIO6 as PWM pin.

kueckermann, 2024-05-20:
Ok thanks. Very helpful. I'll debug this locally on one of my devices.

kueckermann, 2024-05-21:
@benjaminfrancois @Tom-Hirschberger there is an issue in how the PWM line was being requested. I'm working on a patch in the PR: https://github.com/ExplorationSystems/opengpio/pull/11

I'll test this tonight on my local device to confim the fix, and if correct I'll merge and push a new version.

kueckermann, 2024-05-21:
@benjaminfrancois @Tom-Hirschberger the issue has been addressed and deployed to version 1.0.14 on npm. Let me know if you encounter any issues and I'll reopen the ticket.