ExpeditionExploration/opengpio #14: Device Independence

srl295, 2024-08-15:
Hi, first thanks for this.  I was scared away by #1 at first but came back because the solution there (the gpiod2.1) actually works on a pi5.

So, a requests:

It would be nice to have some way to get 'current' instead of hard coding the device in the namespace. Perhaps a function that can get the current device, then a switch based on that device.

`CurrentDevice.output(CurrentDevice.bcm[somePin])`

kueckermann, 2024-08-16:
Hey @srl295, thanks for the suggestion. I've created a separate enhancement issue #15  and will give some thought into how that process can be improved. 

Regarding the second issue. At first glance, it seems the issue with the ts build is that `RaspberryPi_5B.bcm[...]` is expecting the type of the input to be `keyof RaspberryPi_5B['bcm']` but your config at `tankConfig.tank_clk` is likely typed as a string in your config. 

Based on the information you've provided, I'd suggest one of the following solutions:
1. Type `tank_clk` more verbosely so it's strongly typed: 
```
const tank_clk: keyof RaspberryPi_5B['bcm'] = 'GPIO1'; // This should ensure that your values is always a valid key in the bcm map.
```
2. If you find strongly typing it a bit too much, and you're confortable with using the `any` type in your code, then you can do:
```
this.clk = RaspberryPi_5B.output(RaspberryPi_5B.bcm[tankConfig.tank_clk as any]);
```

I've not tested these, so excuse if I've made any errors, but you should get the general idea. Let me know if this addresses the issue. If not, then can you send your file containing `this.clk = RaspberryPi_5B.output(RaspberryPi_5B.bcm[tankConfig.tank_clk]);` and your file containing `tankConfig`.

srl295, 2024-08-16:
@kueckermann yeah, i think I have done similar before.. `tank_clk` is from a config file (imported json), so it's just a string.  

I will try the any, thanks!

srl295, 2024-08-16:
I retitled the issue somewhat to reflect the issue split.

kueckermann, 2024-08-16:
Thanks @srl295, I just saw the reference to the code here: https://github.com/codehivetx/barrelcactus-watermeter/blob/520c11af53ad0e8dd3aaedd24a156a062f66ad21/src/gpio/tankgpio.ts#L173-L177

I think in your case, I'd personally go with using `as any`. However, as part of this issue, I'll look into opportunities to reference the pins with type string.

srl295, 2024-08-16:
> Thanks @srl295, I just saw the reference to the code here: https://github.com/codehivetx/barrelcactus-watermeter/blob/520c11af53ad0e8dd3aaedd24a156a062f66ad21/src/gpio/tankgpio.ts#L173-L177
>
> I think [...]

I'll give this a try.

Meanwhile, super happy that this library has worked so well. The web app and prometheus monitoring are in production. Thanks!

srl295, 2024-08-16:
just a question i wonder could `output()` (and friends) take `Gpio | string` ? I guess not easily since output() is static.

kueckermann, 2024-08-21:
@srl295 good idea and should be doable to make that happen. What do you think if there were 2 ways to use a device? 
1. As a static class - as it is done now.
2. By instantiation:
```js
const rpi = new RaspberryPi_5B()
const gpio4Output = rpi.output('GPIO4')
```

Would something like that work?

srl295, 2024-08-21:
Looks good, with #15 for the discussion of the device itself.

kueckermann, 2024-12-20:
@srl295 I've not forgotten about this. Has been a very busy end of year... home renovation and a 10 month old! Picking this up soon and planning to put it into the version 2 release.

kueckermann, 2025-05-05:
Version 2 is now released (finally). Can be used with `npm install opengpio@latest`.
Version 2 now supports selecting pins by string, number, or the constants as before.
Will be closing this issue now but feel free to open any new issues as they arise.

closed 2025-05-05 by kueckermann