ExpeditionExploration/opengpio #15: Allow easier device referencing to the same device

kueckermann, 2024-08-16:
This request is referenced from #14 

**Origional request:**
It would be nice to have some way to get 'current' instead of hard coding the device in the namespace. Perhaps a function that can get the current device, then a switch based on that device. `CurrentDevice.output(CurrentDevice.bcm[somePin])`

srl295, 2024-08-21:
this could also be a ctor or getter

```js
const rpi = new CurrentDevice()
const rpi = CurrentDevice.getCurrent()
```

then

```js
const gpio4Output = rpi.output('GPIO4')
```

kueckermann, 2025-05-05:
Version 2 is now released (finally). Can be used with `npm install opengpio@latest`.
Will be closing this issue now but feel free to open any new issues as they arise.

With regard to this specific request, I opted not to added anything especially new to support this however there are a few approaches to achieve this already.

Approach 1:
All devices are already static singleton classes so there's no need to instantiate them. You can always use the device directly or you can store the device as a variable.
```ts
// Use directly.
const output = RaspberryPi_5B.output('GPIO4')

// Use variable.
const device = RaspberryPi_5B;
const input = device.input('GPIO4')
```

Approach 2:
My recommendation is to use something similar to Approach 1 but store the device in a file locally and extend the device. That way you can always add your own custom methods.

```ts
// In your local filesystem: device.ts

export default class MyDevice extends RaspberryPi_5B {
    makeCoffee(){
        // Makes coffee
    }
}

// In your application
import device from './device.ts'

device.makeCoffee();
```

I might still make these devices instantiatable in future work, but it won't change how things working ultimately, those instance methods will just wrap to the static methods on the device.

closed 2025-05-05 by kueckermann