ExpeditionExploration/opengpio #16: GPIO4_B2, GPIO4_B3 from OrangePi 5 missing

baybal, 2024-11-07:
GPIO4_B2, GPIO4_B3 from OrangePi 5 missing

kueckermann, 2024-12-17:
Thanks @baybal will aim to fix this week.

jtholmi, 2025-07-04:
Which OrangePi 5 are you @baybal using? Could you provide some schematic or datasheet that shows we are missing GPIO4_B2 (chip: 4, line: 10?), GPIO4_B4 (chip: 4, line: 12?)?

For example, if you were using [OrangePi 5 Plus](https://www.amazon.com/Orange-Pi-Computer-Frequency-Android/dp/B0C9J33MRB) with more GPIO pins, it still doesn't have any GPIO4 pins:
[image]
See [the corresponding schematic on page 11](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-plus.html):
[image]

jtholmi, 2025-07-04:
Do you have [OPi 5 Pro](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-Pro.html), for which Schematic page 25 shows GPIO4_B2, GPIO4_B3?

[image]

baybal, 2025-07-05:
I had Opi 5 plus 
[image]