ExpeditionExploration/opengpio #23: SyntaxError: The requested module 'opengpio' does not provide an export named 'RaspberryPi_5B'

benediktdertinger, 2025-01-22:
Hi there, if I execute the code below I get the error shown afterwards. I use `tsx` for compilation and other lib imports work as expected. Not sure if this is related to my setup. Do you have any idea?

I'm using:
```
Node: v20.12.2 and ES6 Modules

"main": "index.js",
"type": "module",
"scripts": {
  "start": "tsx src/index.ts",
},
...
"dependencies": {
  "opengpio": "^1.0.17"
},
"devDependencies": {
  "tsup": "^8.3.5",
  "tsx": "^4.19.2",
  "typescript": "^5.7.3"
}
```

Import and execution:
```ts
import { RaspberryPi_5B } from 'opengpio';

const pwm = RaspberryPi_5B.pwm(RaspberryPi_5B.bcm.GPIO18, 0, 333);

```

ERROR:
```
import { RaspberryPi_5B } from 'opengpio';
         ^

SyntaxError: The requested module 'opengpio' does not provide an export named 'RaspberryPi_5B'
    at ModuleJob._instantiate (node:internal/modules/esm/module_job:134:21)
    at async ModuleJob.run (node:internal/modules/esm/module_job:217:5)
    at async ModuleLoader.import (node:internal/modules/esm/loader:323:24)
    at async loadESM (node:internal/process/esm_loader:28:7)
    at async handleMainPromise (node:internal/modules/run_main:113:12)
```

If I run it like this, it works:
```ts
import gpio from 'opengpio';

const pwm = gpio.RaspberryPi_5B.pwm(gpio.RaspberryPi_5B.bcm.GPIO18, 0, 333);

kueckermann, 2025-03-21:
Thanks @benediktdertinger for letting us know. I'll take a look at why might be happening.

kueckermann, 2025-05-05:
@benediktdertinger this should be fixed now in Version 2.
------
Version 2 is now released (finally). Can be used with npm install opengpio@latest.
Will be closing this issue now but feel free to open any new issues as they arise.

closed 2025-05-05 by kueckermann