ExpeditionExploration/opengpio #25: Unable to catch errors

MeestorX, 2025-01-28:
If there's an error in opengpio (for example, if you try to run a function that's not supported by the hardware), I have not been able to try/catch the error, it simply crashes the software. Obviously not ideal!

Is there a way to have the program work with try/catch blocks?

e.g. This will crash the entire program when I'm developing on another device that doesn't support the hardware.

```
try {
         watch = Radxa_Rock_S0.watch(encoderPin, Edge.Falling);
} catch (err) {
         console.log('watch could not be created!')
}
```
The error it produces:
```
terminate called after throwing an instance of 'std::system_error'
  what():  cannot open GPIO device gpiochip2: No such file or directory
Aborted
error Command failed with exit code 134.
```

kueckermann, 2025-03-21:
Hey @MeestorX, I'll take a look at this soon. We've finally solved the install problems in version 2. So we're working to finalise some of the other issues and we'll release that version within the next few weeks.

kueckermann, 2025-07-04:
@MeestorX can you test if you can reproduce this error anymore in version 2 now that it's released?