ExpeditionExploration/opengpio #26: abstract type definition to platform-agnostic package

2bndy5, 2025-03-07:
When developing a Linux project from a non-Linux machine (eg Windows), it would be helpful to be able use the type definitions without having libgpiod-dev installed.

Looking at the source, I'm not sure how plausible this idea is. Just thought I'd ask anyway.

Currently, I have to add this package as an optional dependency and use WSL (with libgpiod-dev installed) just to get the type definitions available in Windows.

kueckermann, 2025-03-19:
Thanks for the idea. I've actually been working on this in the new version of opengpio which is currently in progress 👍

kueckermann, 2025-05-05:
Version 2 is now released (finally). Can be used with npm install opengpio@latest.
Will be closing this issue now but feel free to open any new issues as they arise.

You can now find in the **Local Development** section of the README details on using opengpio in mocked mode, but in a nut shell:
- Install without scripts `npm i --save opengpio --ignore-scripts`. This prevents the build from running, which would have otherwise fail if you are not on linux with the relevant dependancies installed.
- Run in mocked mode by setting OPENGPIO_MOCKED=true environment variable. This will cause opengpio to not load native bindings and instead set all bindings to mocked functions. You can override the mocks with your own functions if you want, by detecting the mock and replacing the methods as you wish. By default the mocks will always return true for pin state and use blank functions for pin setters and cleanup.

```ts
// When running with OPENGPIO_MOCKED=true environment variable.
import opengpio from 'opengpio';
if (opengpio.mocked) {
    console.log('opengpio is running with mocked bindings');
    // Replace core bindings if you want. Eg.
    opengpio.bindings.input = (chip: number, line: number, bias: number) => [
        ()=> true, // Return the pin state that you want
        ()=>{} // Cleanup function
    ]
}
```

closed 2025-05-05 by kueckermann