ExpeditionExploration/opengpio #34: Installing unstable debian package source and installing dependencies bricks Raspberry Pi OS Desktop Environment

Matix-Media, 2025-05-20:
Installing the Debian unstable package source (sid) and installing any dependency after doing so bricks the whole desktop with a Raspberry Pi.

When installing the deps, I get prompted to almost remove all dependencies and install new ones from the new source, in the process a lot of configs get overridden and some crucial deps for desktop environment are removed completely without a replacement. Is there a better solution to use v2 of libgpiod-dev or do I have to relay on v1 (which means I have to use opengpio 1)?

2bndy5, 2025-05-20:
I believe [libgpiod can be installed from source](https://libgpiod.readthedocs.io/en/latest/building.html) as long as you are using an updated Linux kernel.

> [!tip]
> libgpiod v2 requires the Linux kernel's GPIO character device ABI v2. libgpiod v1.x only used the Linux kernel's GPIO character device ABI v2.

The Linux kernel's GPIO character device ABI v2 has been deployed along side ABI v1 for a while, but I'm not sure what exact version of the Linux Kernel deployed ABI v2.

kueckermann, 2025-05-21:
@jtholmi any thoughts here?

Matix-Media, 2025-05-22:
I've successfully built `libgpio 2.2.1` locally but when I run `pnpm add opengpio` there is seemingly no postinstall script executed and when I run a test script:
```ts
import { Default } from 'opengpio';

(async () => {
const output = Default.output({chip: 0, line: 4});
output.value = true; // Set the RaspberryPi 5B's GPIO14 pin high
await new Promise((resolve) => setTimeout(resolve, 2000));
output.value = false; // Set the RaspberryPi 5B's GPIO14 pin low
})();
```
I get the following error:
```
Error: Could not locate the bindings file. Tried:
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/Debug/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/Release/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/out/Debug/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/Debug/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/out/Release/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/Release/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/default/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/compiled/22.15.1/linux/arm64/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/release/install-root/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/debug/install-root/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/default/install-root/opengpio.node
 → /home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/lib/binding/node-v127-linux-arm64/opengpio.node
    at bindings (/home/nicks/software/test/node_modules/.pnpm/bindings@1.5.0/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/src/bindings.ts:11:23)
    at Module._compile (node:internal/modules/cjs/loader:1730:14)
    at Object.transformer (/home/nicks/.cache/pnpm/dlx/9eef885f6593e10f151bcf0c0a5b19c79f7a9d50ae7339d9a404279f34f46b02/196f8a90084-2439/node_modules/.pnpm/tsx@4.19.4/node_modules/tsx/dist/register-D2KMMyKp.cjs:2:1186)
    at Module.load (node:internal/modules/cjs/loader:1465:32)
    at Function._load (node:internal/modules/cjs/loader:1282:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
    at Module.require (node:internal/modules/cjs/loader:1487:12)
    at require (node:internal/modules/helpers:135:16) {
  tries: [
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/Debug/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/Release/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/out/Debug/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/Debug/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/out/Release/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/Release/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/build/default/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/compiled/22.15.1/linux/arm64/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/release/install-root/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/debug/install-root/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/addon-build/default/install-root/opengpio.node',
    '/home/nicks/software/test/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/lib/binding/node-v127-linux-arm64/opengpio.node'
  ]
}
```
> [!note]
> I have opengpio listed in `pnpm-workspace.yaml` under `onlyBuiltDependencies` so that pnpm does not skip the build of the lib.

kueckermann, 2025-05-22:
@Matix-Media Can you try uninstall opengpio from your project and reinstall it and see if build runs. Alternatively you could try cd into node_modules/opengpio and run `npm run build` or whatever the pnpm equivilant is. No sure why pnpm would run the build script. If you know, then let me know and I can update the package if specific changes are needed for pnpm support.

2bndy5, 2025-05-22:
> The latest Raspberry Pi OS release uses kernel version 6.12.

That should be sufficient. I have another C++ project that directly uses the Linux kernel's GPIO character device ABI v2 (without libgpiod), and it builds/runs fine with that kernel version.

2bndy5, 2025-05-22:
I'm not sure if this information helps. The following is an exploration of my RPi3 (64-bit running RPi OS).
<details><summary><code>sudo apt list *gpiod*</code></summary>
<p>

I stripped away the non-relevant packages (like `pigpio`) for brevity.

```
Listing... Done
gpiod/stable,now 1.6.3-1+b3 arm64 [installed]
gpiod/stable 1.6.3-1+b3 armhf
libgpiod-dev/stable,now 1.6.3-1+b3 arm64 [installed]
libgpiod-dev/stable 1.6.3-1+b3 armhf
libgpiod-doc/stable,stable 1.6.3-1 all
libgpiod2/stable,now 1.6.3-1+b3 arm64 [installed,automatic]
libgpiod2/stable 1.6.3-1+b3 armhf
python3-libgpiod/stable,now 1.6.3-1+b3 arm64 [installed]
python3-libgpiod/stable 1.6.3-1+b3 armhf
rgpiod/stable 0.2.2-1~rpt1 arm64
rgpiod/stable 0.2.2-1~rpt1 armhf
```

</p>
</details> 

<details><summary><code>dpkg -L libgpiod2</code></summary>
<p>

```
/.
/usr
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/libgpiod.so.2.2.2
/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1.1.2
/usr/share
/usr/share/doc
/usr/share/doc/libgpiod2
/usr/share/doc/libgpiod2/changelog.Debian.arm64.gz
/usr/share/doc/libgpiod2/changelog.Debian.gz
/usr/share/doc/libgpiod2/copyright
/usr/lib/aarch64-linux-gnu/libgpiod.so.2
/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1
```

</p>
</details> 

<details><summary><code>dpkg -L libgpiod-dev</code></summary>
<p>

```
/.
/usr
/usr/include
/usr/include/gpiod.h
/usr/include/gpiod.hpp
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/libgpiod.a
/usr/lib/aarch64-linux-gnu/libgpiodcxx.a
/usr/lib/aarch64-linux-gnu/pkgconfig
/usr/lib/aarch64-linux-gnu/pkgconfig/libgpiod.pc
/usr/lib/aarch64-linux-gnu/pkgconfig/libgpiodcxx.pc
/usr/share
/usr/share/doc
/usr/share/doc/libgpiod-dev
/usr/share/doc/libgpiod-dev/README.gz
/usr/share/doc/libgpiod-dev/changelog.Debian.arm64.gz
/usr/share/doc/libgpiod-dev/changelog.Debian.gz
/usr/share/doc/libgpiod-dev/copyright
/usr/lib/aarch64-linux-gnu/libgpiod.so
/usr/lib/aarch64-linux-gnu/libgpiodcxx.so
```

</p>
</details>

There is no `libgpiod2-dev` package available from RPi OS's default PPAs.

<details><summary><code>sudo apt search libgpiod*</code></summary>
<p>

```
Sorting... Done
Full Text Search... Done
gpiod/stable,now 1.6.3-1+b3 arm64 [installed]
  Tools for interacting with Linux GPIO character device - binary

libgpiod-dev/stable,now 1.6.3-1+b3 arm64 [installed]
  C library for interacting with Linux GPIO device - static libraries and headers

libgpiod-doc/stable,stable 1.6.3-1 all
  C library for interacting with Linux GPIO device - library documentation

libgpiod2/stable,now 1.6.3-1+b3 arm64 [installed,automatic]
  C library for interacting with Linux GPIO device - shared libraries

python3-libgpiod/stable,now 1.6.3-1+b3 arm64 [installed]
  Python bindings for libgpiod (Python 3)
```

</p>
</details>

---
## Steps to install `libgpiod` v2 from source

> [!IMPORTANT]
> First, we need to uninstall any libgpiod v1 from apt packages
> ```sh
> sudo apt remove libgpiod-dev gpiod libgpiod2
> ```
> This will also uninstall `python3-libgpiod`. But users can still install the latest [`gpiod` package from PyPI](https://pypi.org/project/gpiod/) when needed.

```bash
# to avoid typos, we'll use a bash variable.
LIBGPIOD_VERSION=2.2.1
# published package versions are found by browsing
#   https://mirrors.edge.kernel.org/pub/software/libs/libgpiod/
# for opengpio v2, we just need to make sure we are using 2.y.z

wget https://mirrors.edge.kernel.org/pub/software/libs/libgpiod/libgpiod-${LIBGPIOD_VERSION}.tar.xz
tar -xvf ./libgpiod-${LIBGPIOD_VERSION}.tar.xz
cd ./libgpiod-${LIBGPIOD_VERSION}/
./configure --enable-tools --enable-bindings-cxx
make
sudo make install
```
Afterwards, you should now be able to use `opengpio` v2.x in any node.js project located on that system.

Matix-Media, 2025-06-06:
Sorry for the re-open, but I was fooled, building the bindings worked flawlessly, but when I try to actually run my transpile (ts -> js) code, I get the following error:
```
/home/nicks/software/windrad-bsync/node_modules/.pnpm/bindings@1.5.0/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: libgpiodcxx.so.2: cannot open shared object file: No such file or directory
    at Object..node (node:internal/modules/cjs/loader:1921:18)
    at Module.load (node:internal/modules/cjs/loader:1465:32)
    at Function._load (node:internal/modules/cjs/loader:1282:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
    at Module.require (node:internal/modules/cjs/loader:1487:12)
    at require (node:internal/modules/helpers:135:16)
    at bindings (/home/nicks/software/windrad-bsync/node_modules/.pnpm/bindings@1.5.0/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/nicks/software/windrad-bsync/node_modules/.pnpm/opengpio@2.0.0/node_modules/opengpio/dist/cjs/bindings.js:15:58)
    at Module._compile (node:internal/modules/cjs/loader:1730:14) {
  code: 'ERR_DLOPEN_FAILED'
}

Node.js v22.15.1
 ELIFECYCLE  Command failed with exit code 1.
```

I don't really know what the problem is here.. I don't think the reason is that I use pnpm.. Since another native bindings library that I use in the same projects works fine.

closed 2025-06-06 by Matix-Media